mrbeam/OctoPrint #8: Image to .gco

only1pj, 2015-02-02:
# Feature Request
# Enhancment

Can the tool used within the web interface be turned into a stand alone application to have the host machine "slice" the image instead of the raspberry pi? Much faster and more reliable.

Some images are taking hours and not generating anything, the only way to end the process is to power cycle the Mr Beam and try again or give up.

The InkScape addon has not worked once for me yet. Even with images that have worked in the web interface.

hungerpirat, 2015-02-03:
@only1pj I am currently working on that topic. 
The slicing code on the raspberry and the inkscape extension are the same. Additionally it can be called via command line on the host computer. python mrbeam.py --help shows you all the options.
There are a lot of bugs with svg graphics with certain namespaces for example. 
I am currently working on these issues. Expect an update soon.

only1pj, 2015-02-03:
Thanks Teya,
I knew the extension was meant to be the same, I coudnt understand why it has never worked. 
Each time I run it I get the following error:
"This application has requested the Runtime to terminate it in an unusual way. Please contact the applications support team for more information."
then
"Inkscape encountered and internal error and will close now. Automatic backups of unsaved documents were done to the following locations
C:\uers\etc\etc..."
Inkscape then crashes and closes.

It does create a .nc file where I asked it to but there are only a few lines, which just looks like a default start to a file.

$H
G92X0Y0Z0
G90
M08
G21

M05S0
G0 X0.000 Y0.000
M09
M02

I tried running it from the command line on a windows machine and got this:

Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Users\PJ>python mrbeam.py --h
'python' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\PJ>

I dont know where the python executable lives so i'll leave it alone, 

Looking forward to the update!
Thanks for your work so far!!

PJ

mrbeam, 2015-02-03:
Oh - sorry, I should have written some more explanation about python.
Python is the programming language the extension is written in.
Though it is pre-installed on most linux systems, you have to install it on
windows manually. Checkout https://www.python.org/downloads/windows/ and
install the latest python 2 release.
Probably you will need to install some additional python libraries - but I
can't give you a hint how to approach here as I am not familiar with
windows. Sorry for that.
Maybe someone on our mailing list knows how to do that.

best, teja

-- 
--------------------------------------
Teja Philipp
Mr Beam Lasers LLC

6008 N Lamar Blvd., #214

78752 Austin TX

Mr Beam Lasers UG (haftungsbeschränkt)
Elvirastr. 11, 80636 München
Geschäftsführer: Teja Philipp
Handelsregister: Amtsgericht München, HRB 212375

make-ing, 2016-06-21:
Hello only1pj,
is this issue still valid? We currently think about moving the gcode generation to the front end (host machine) but tests with the new RPi3 are showing a significant boost in performance. So its not sure if we switch.

greetings Florian

only1pj, 2016-06-21:
Hi,
It's been. While since I was able to user Beam but I would say the issue
would still remain for people who do not wish to buy another RPi,
disconnect, replace, and reconnect everything again.
The host machine will always be the most powerful and fastest method of
doing this and should lower bandwidth requirements if the code stayed on
the host instead of being created on the pi and being sent to the host.

make-ing, 2016-06-22:
@only1pj: The Host machine will indeed be the more powerful machine in most cases, but it actually requires more bandwidth. You still have to send the generated Gcode to the back-end which is plane text and can get very big in no time compared to the image files you would send to the back-end. 

I leave this issue open until we have a decision on this. Thanks for your comments.