mrbeam/OctoPrint #10: Storage Area

only1pj, 2015-02-08:
~enhancement

Is it possible to access the storage area from a windows/linux/other computer?
Access to the storage area would allow for easy backup or restore of files if the SD card needed to be re-imaged for example. 

Currently unable to map a drive on a windows pc to the Raspberry Pi. Dont know which ports are open, which services are installed/running, what the login details would be.

This would be useful while software/firmware updates are being worked on. I for one do not want to have to find all the files I have already uploaded and spent hours "slicing" before printing them.

PJ

hungerpirat, 2015-02-09:
Interesting idea.
This would make sense only with a cross platform protocol - maybe ftp or
nfs?
Alternatively you could use a dropbox / owncloud / google drive client?
I am not sure which solution will consume how much cpu power on the
raspberry.
As far as I know, the files api in OctoPrint is pretty much REST style -
are there any tools to mount a REST accessible folder on another machine ?

good thing is: this does not affect the webinterface code at all - it can
be done separately.
just thoughts...

teja

make-ing, 2016-06-21:
Hello PJ,
i don't think this is necessary anymore, because updates now don't force you to re-flash the SD-card. Maybe we add a "Download all Files" Button in the settings menu later to allow you to quickly download all stored files.

greetings Florian