mrbeam/OctoPrint #11: Chrome Web Interface on Mac

RealGaryBall, 2015-02-21:
I am having trouble viewing the entire web interface when using Google Chrome on my Macbook.  I am unable to scroll to the top of the window and miss the top bit.  The top set of menus are covering up the middle window. 

I have the same problem when I use Safari but I can temporarily scroll the window down to see what I am missing but it bounces right back up when I let go of the scroll.

[image]

P.S.  The update looks great and I am excited to put it to use.  It appears as if it will clear up many of my issues with the previous version.

hungerpirat, 2015-02-21:
Hi Gary,

thanks for the compliments about the update.

I probably did not run into this issue because my login "teja" is way
shorter than your username "real.gary.ball" on the right side of the menu
bar.

Can you tell me the display resolution of your macbook ?

~teja

RealGaryBall, 2015-02-21:
1280x800

Is it possible to turn off the sliders for laser intensity and speed?  I preferred entering the numbers by hand.

only1pj, 2015-02-21:
If you add your own template size thought the options/settings then the
values on the sliders change. Ive seen from 2000 to something in the
hundreds, up to 5000.

commit 43a476ed (2015-03-01): fixes #11
closed 2015-03-01 by hungerpirat