mrbeam/OctoPrint #14: SVG created from online editor does not work

mrbeam, 2015-03-17:
svg file from an online editor does not work. 

'''xml
<svg xmlns="http://www.w3.org/2000/svg" xmlns:svg="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" width="100%" height="100%" viewBox="0 0  1066 753.0138854980469" preserveAspectRatio="xMidYMid meet" zoomAndPan="disable">
<defs id="svgEditorDefs">
<polygon id="svgEditorPolygonDefs" stroke="black" fill="khaki" style="vector-effect: non-scaling-stroke; stroke-width: 1px;"/>
</defs>
<rect id="svgEditorBackground" x="0" y="0" width="1066" height="753.013916015625" style="fill: none; stroke: none;"/>
<circle id="e1_circle" cx="200" cy="200" stroke="black" style="stroke-width: 1px; vector-effect: non-scaling-stroke;" r="4mm" fill="khaki"/>
<circle id="e2_circle" cx="200" cy="200" style="stroke-width: 1px; vector-effect: none; fill-opacity: 0;" r="10mm" stroke="black"/>
</svg>
'''

make-ing, 2016-06-17:
first problem was the missing xml tag before the svg tag. which is now fixed in the dev branch with commit 25be525. Second problem is with the units in mm in the svg attributes. This still needs to be fixed

make-ing, 2016-06-17:
which online editor was used to produce this file?

make-ing, 2016-06-17:
fixed problem with the mm units in the svg elements by converting them to px: 12a3f092
This update will be in the next stable release.

closed 2016-06-17 by make-ing