mrbeam/OctoPrint #29: manually filled areas have gaps

mrbeam, 2015-06-16:
[image]

whatever may be the reason.
First step: check gcode creation and repeatability.

Dregbaer, 2015-06-18:
[image]
It seems like it isn't skipping paths, but putting them at wrong places

only1pj, 2015-06-18:
I used this cookie cutter method to generate a SVG and although it did run for a few hours, I knocked the material while it was burning so cancelled the job. 
I didnt have any issues with missing areas.
That one in the ladt image looks quite bad.

hungerpirat, 2015-06-18:
Did you check if your axes are loose or the timing belt has backlash?
It looks like this happens if the axis orthogonal to the filling lines
changes direction.

Can you confirm that?

Dregbaer, 2015-06-18:
Indeed my axes were a bit loose and the timing belt was scratching at the plastic parts. Fixed this and tried to prnt again. It made some mistakes at the beginning, 
[image]
but then printed everything where it belonges.
[image]
I'll run some more tests, but already consider my problem solved.
So thanks for helping me!

mrbeam, 2015-06-18:
You're welcome.

try to loosen the maden screws of the pulleys that they can freely spin
over the motor shaft.
Then move the gantry along the axes and try to feel if there is different
friction in some areas.
If yes, look at the rods , check the linear bearings and their seals, maybe
put a drop oil on the rods.
That should help.

mrbeam, 2015-08-31:
another hint: 
linear bearings with dry running sealings cause quite some backlash which results in this effect.
-> lubricate the rods.

closed 2015-08-31 by mrbeam