mrbeam/OctoPrint #36: Software lagging command execution

mrbeam, 2015-06-30:
Hi all,

I have the impression that the software GUI (OctoPi) is lagging in sending commands to the arduino.
If I click a button it takes sometimes 1-3 seconds before the command gets executed.

I'm on the latest update via the GUI.

only1pj, 2015-07-04:
Plus one for this.
Its not a new issue - had this from day one. 
Seems more of an issue when just booting or just before/after a long process.

make-ing, 2016-06-21:
This seems to be a problem with the general speed of the Raspberry Pi 1. During boot and short after this the pi is still under heavy load due to all the services starting up and the connections are established. Same holds true for long processes. This will be solved a soon as we switch to RPi3.

closed 2016-06-21 by make-ing