mrbeam/OctoPrint #45: Feature Request: Quick Text

only1pj, 2015-09-16:
With the recent introduction of the built in grey scale test image, it would be a nice idea to be able to quickly burn the settings used to a piece of test material.

eg, Burn test piece. Fails. Adjust settings. Burn again with different setting > OK. 
"Quick text" would be a simple text box for "Speed, Intensity, Type of Material" that can be typed in the web interface and burned directly to the material instead of having to use inkscape to produce file and have it converted first. This lets you keep stock for quick reference as to what materials was used and what the output looked like. Saves trial and error each time.

Could be further enhanced to produce "quick engraving" on objects. 
eg, Enter in to web interface "text", choose font, choose size, burn to object.

TopperDEL, 2015-09-16:
What a nice idea! I like it!

mrbeam, 2015-09-16:
Great! 
I have to think about a font rendering solution either in the browser or on the RPi . 
Maybe a screenshot with js could be done and then I am not in charge of any font conversation.

only1pj, 2015-09-16:
I cant help feeling you said "Great" through gritted teeth ;)

A generic none copyright system font would do for the "quick text". Nothing special needed unless it did evolve into a quick engrave tool. I saw the 0% and 100% already on the greyscale. Maybe just use the same type of code used to pre-populate the file name when converting?

Print greyscale. Tickbox to print current settings beneath it or not. :)

amrsoll, 2020-01-09:
Feature now available

closed 2020-01-09 by amrsoll