mrbeam/OctoPrint #46: Visual progress of gco files

only1pj, 2015-09-18:
I added the greyscale image to the work area, had to resize it slightly to fit my material and clicked convert. Didnt change settings as it was first test.
On clicking convert there was no confirmation it had finished. I clicked again. Nothing. Closed the page thinking it had failed. There were two .gco files in the design library.

I had removed the png file from the work area but clicked add on one of the gco files.
It worked; currently burning as I type this, but the gco file which was shades of red when added, no longer shows on the work area so cannot see how far through the burn process it is.

only1pj, 2015-09-18:
Convert for images appears to finish but not notify that it's done.
Refresh page and accept agreement, go to library and .gco file is there.

only1pj, 2015-09-22:
Latest version (cant check as burning image and dont want to mess it up) of octoprint now rotates images but gcode image still does not show when burning.

make-ing, 2016-06-21:
@only1pj: The behavior that the gcode (red shades) is not showing anymore during burning is normal. But the behavior that it's not notifying when its done is not normal. Do you still have this problem? Otherwise i will close this issue soon.