mrbeam/OctoPrint #47: Progress update (% bar) freeze

only1pj, 2015-09-22:
Last test image I produced, the progress bar stopped at 30+%. Only just cleared now after about 5 mins after the burn finished. There was a long pause during the burn 30/40 seconds(?). Thankfully the laser was not on during the pause as may have caused damage.

Maybe the progress bar stopped at the same time but I wasnt looking.

tylirw, 2015-10-02:
The slicing bar freezes at 66% and remains that way until it is finished slicing (which now takes significantly longer than before the most recent update) or until I unplug the laser and restart the system. Rebooting and/or restarting Octoprint does nothing. In addition, I can no longer slice multiple files at once. Being able to hide the slicing process and start a new slicing process at the same time was a serious advantage of the old software.

The laser has also been very slow since last update. The files I have been using are not images, just plain svgs. The laser pauses frequently and during these pauses the laser seems to be running at low intensity. Engraving the same svgs I used to before the new update is much slower after the new slicing process (I re-sliced to test the difference).

mrbeam, 2015-11-25:
to be tested with the latest release.

make-ing, 2016-06-16:
I think these are two different things.
-The progress bar during lasering is stopping as soon as the lasers stops, this is normal behavior.
-The progress bar during slicing is not very granular so it seems that is freezes but it just takes a longer time from 66% to 100% then from 0% to 66%.

If you still have problems please write a comment and i will reopen this issue. Until then, invalid.

closed 2016-06-16 by make-ing