mrbeam/OctoPrint #48: Laser movement issues (freeze)

only1pj, 2015-09-22:
While burning the greyscale test image, the laser head, while ON, would pause in one place burning into the material more than it should.
This, depending on the cause, could cause issues (possible fire hazard) on long jobs if unattended for any reason. Even if attended, the user should not be staring at the laser light for any length of time even with glasses.

Network/adapter/connection?
Pi?
Cache/RAM speed?

only1pj, 2015-10-07:
Just had another instance of this. While testing some vector cuts after the new update (buffer under run) to fix dithering raster images, the machine ran a few jobs fine, re-converting the same SVG each time with different speed/intensity settings.

The laser started, drew one shape, approx 100mm circumference, started another and moved only 30mm before the head stopped and the laser stayed on. The laser cut 1 or 2 mm into the MDF bed it was resting on.

make-ing, 2016-06-21:
@only1pj: Can you give me the last 20 or so lines form the terminal tab when this problem happens?