mrbeam/OctoPrint #52: Dwel where there shouldn't be

only1pj, 2015-10-07:
Each outline shows where the laser paused before starting the shape. NO dwel set on gcode. Could not attach so pasted at the bottom.

[image]

This was a good one - Paused until I pressed cancel on the machine. Drilled 2mm into the MDF under it.
[image]

;Generated from UnionFlag_i1000s750.svg e2f7fbb9847e3a643d99c548eedc62e0ffef5cca
G21

; Layer: layer1, outline of path3374
G0 X32.3603 Y23.0027

M03 S1000
G01 F750.000000
G01  X62.1286 Y44.5182
G01  X68.5969 Y44.4210
G01  X75.0652 Y44.3239
G01  X45.3473 Y22.8839
G01  X15.6294 Y1.4444
G01  X9.1107 Y1.4729
G01  X2.5920 Y1.5013
G01  X32.3603 Y23.0167
G01  X32.3603 Y23.0027
M03 S0

G0 X22.8673 Y1.5907

M03 S1000
G01F750.000000
G02 X23.0538 Y1.7772 I0.6745 J-0.4883
G02 X49.8882 Y21.1564 I15280.7966 J-21131.0834
G02 X76.7693 Y40.5174 I15247.0610 J-21140.8938
G01 X76.8991 Y40.5592
G01 X76.9619 Y40.4968
G02 X77.0243 Y21.0035 I-3044.2096 J-19.4933
G01  X77.0243 Y1.4479
G01  X49.8940 Y1.4479
G02 X22.9025 Y1.5205 I0.0000 J5015.2727
G01 X22.8673 Y1.5907
G01  X22.8673 Y1.5907
M03 S0

G0 X2.4062 Y30.2432

M03 S1000
G01 F750.000000
G01  X2.4689 Y44.3273
G01  X22.0394 Y44.4130
G02 X41.1621 Y44.3124 I4.3698 J-987.2101
G01 X41.2372 Y44.1270
G02 X40.6574 Y43.6327 I-3.4577 J3.4680
G02 X21.7517 Y29.9385 I-5281.0289 J7270.9202
G02 X2.7086 Y16.2204 I-5249.8975 J7267.6896
G02 X2.5209 Y16.1600 I-0.1877 J0.2612
G01 X2.4321 Y16.2481
G02 X2.4062 Y30.2438 I1111.0490 J9.0522
G01  X2.4062 Y30.2432
M03 S0

G0 X2.3929 Y101.9600

M03 S1000
G01F750.000000
G02 X2.4793 Y115.9830 I2661.8361 J-9.3743
G01 X2.5172 Y116.0946
G01 X2.6048 Y116.1079
G02 X22.1010 Y102.0788 I-2694.5571 J-3765.1916
G01  X41.6099 Y87.9639
G01  X21.9767 Y87.9353
G01  X2.3435 Y87.9068
G01  X2.3929 Y101.9423
G01  X2.3929 Y101.9600
M03 S0

G0 X25.7046 Y104.4729

M03 S1000
G01F750.000000
G02 X2.8736 Y121.0278 I5066.8027 J7011.7031
G02 X2.6674 Y121.2348 I0.5451 J0.7492
G02 X2.5186 Y121.6649 I0.6387 J0.4618
G02 X2.4110 Y126.0783 I140.0626 J5.6201
G02 X2.4461 Y130.4778 I132.7362 J1.1427
G01 X2.5867 Y130.6047
G02 X2.8166 Y130.5307 I-0.0000 J-0.3939
G02 X32.2904 Y109.2742 I-6651.7716 J-9254.2470
G01  X61.7402 Y87.9639
G01  X55.1544 Y87.9639
G01  X48.5686 Y87.9639
G01  X25.7046 Y104.4860
G01  X25.7046 Y104.4729
M03 S0

G0 X49.8112 Y111.2028

M03 S1000
G01 F750.000000
G01  X22.7223 Y130.8036
G01  X49.7893 Y130.8950
G02 X76.8217 Y130.9257 I20.1221 J-5824.5878
G01 X76.9425 Y130.8607
G01 X76.9856 Y130.7227
G02 X76.9642 Y111.1682 I-4442.6716 J-4.9166
G01  X76.8999 Y91.6000
G01  X49.8111 Y111.2014
G01  X49.8112 Y111.2028
M03 S0

G0 X107.0954 Y111.2485

M03 S1000
G01 F750.000000
G01  X107.0954 Y130.9839
G01  X134.1843 Y130.9839
G02 X161.1877 Y130.8928 I-0.0000 J-4001.0672
G01 X161.2731 Y130.8067
G01 X161.1997 Y130.6625
G02 X134.1843 Y111.0711 I-4621.3071 J6344.1064
G01  X107.0954 Y91.5148
G01  X107.0954 Y111.2505
G01  X107.0954 Y111.2485
M03 S0

G0 X138.7276 Y109.4793

M03 S1000
G01 F750.000000
G01  X168.4959 Y130.9950
G01  X174.9907 Y130.8979
G01  X181.4856 Y130.8007
G01  X151.7852 Y109.3610
G01  X122.0848 Y87.9216
G01  X115.5221 Y87.9502
G01  X108.9593 Y87.9787
G01  X138.7276 Y109.4944
G01  X138.7276 Y109.4793
M03 S0

G0 X161.3253 Y87.8382

M03 S1000
G01 F750.000000
G01  X142.3656 Y87.9325
G01  X160.0827 Y100.7165
G01  X179.7259 Y114.8917
G01  X181.6519 Y116.2829
G01  X181.6519 Y102.1460
G01  X181.6519 Y88.0093
G01  X180.9685 Y87.8807
G02 X180.2791 Y87.8151 I-0.6976 J3.6711
G02 X161.3253 Y87.8407 I-5.8856 J2649.7604
G01  X161.3253 Y87.8382
M03 S0

G0 X161.7702 Y30.3346

M03 S1000
G01 F750.000000
G01  X142.2612 Y44.4684
G01  X161.9566 Y44.4970
G01  X181.6519 Y44.5256
G01  X181.6519 Y30.3524
G02 X181.5592 Y16.2816 I-1068.3716 J0.0000
G01 X181.4655 Y16.1986
G01 X181.2967 Y16.2530
G02 X161.7701 Y30.3524 I4427.1902 J6151.8867
G01  X161.7702 Y30.3346
M03 S0

G0 X151.8293 Y22.9770

M03 S1000
G01 F750.000000
G01  X122.1310 Y44.4670
G01  X128.7794 Y44.4956
G01  X135.4279 Y44.5241
G01  X158.5399 Y27.8386
G01  X181.6519 Y11.1529
G01  X181.6519 Y6.3091
G02 X181.6210 Y1.5079 I-373.2942 J0.0000
G01 X181.5898 Y1.4853
G01 X181.5336 Y1.5035
G02 X151.8293 Y22.9950 I181953.8166 J251516.7074
G01  X151.8293 Y22.9770
M03 S0

G0 X107.0954 Y21.0035

M03 S1000
G01F750.000000
G02 X107.1882 Y40.4563 I2038.4592 J0.0000
G01 X107.2818 Y40.5395
G01 X107.4506 Y40.4850
G02 X134.4328 Y21.0035 I-8390.7805 J-11649.8053
G01  X161.3974 Y1.4873
G01  X134.2464 Y1.4589
G01  X107.0954 Y1.4301
G01  X107.0954 Y20.9861
G01  X107.0954 Y21.0035
M03 S0

G0 X82.9888 Y27.2820

M03 S1000
G01 F750.000000
G01  X82.9888 Y53.1161
G01  X42.6817 Y53.2075
G01  X2.3745 Y53.2990
G01  X2.2988 Y66.1270
G01  X2.2231 Y78.9551
G01  X42.6043 Y78.9551
G01  X82.9855 Y78.9551
G01  X83.0493 Y104.8798
G01  X83.1131 Y130.8047
G01  X91.9961 Y130.8990
G01  X100.8791 Y130.9933
G01  X100.9429 Y105.0638
G01  X101.0066 Y79.1345
G01  X141.3293 Y79.0431
G01  X181.6519 Y78.9516
G01  X181.6519 Y66.0353
G01  X181.6519 Y53.1187
G01  X141.2671 Y53.1187
G01  X100.8824 Y53.1187
G01  X100.8824 Y27.2831
G01  X100.8824 Y1.4482
G01  X91.9356 Y1.4482
G01  X82.9888 Y1.4482
G01  X82.9888 Y27.2823
G01  X82.9888 Y27.2820
M03 S0

M05

make-ing, 2015-10-12:
I made a workaround by removing the status report messages during printing. But this issue needs more attention.

make-ing, 2015-11-25:
Fixed with the new communication interface.

closed 2015-11-25 by make-ing