mrbeam/OctoPrint #54: Design library gets rendered only partly

TopperDEL, 2015-11-23:
There seems to be an issue within the latest version. The design library is cut off on top and bottom, so I can't see the full list. It show wrong in IE, Edge and Chrome.
[image]

TopperDEL, 2015-11-24:
I realized this only happens when I'm logged in.

mrbeam, 2015-11-25:
damn, this is ugly. Reason: your username is too long and causing the whole menu to wrap to the next line. 
Workaround until fix:  use a shorter username.

TopperDEL, 2015-11-25:
I think I go for "Mr.T"... :)

make-ing, 2016-06-16:
I can't reproduce it, and i think this was fixed with the commit 43a476e. If you still have problems please comment again and i will reopen this issue. Until then, fixed!

make-ing, 2016-06-20:
The problem is still present and seems to be font related. Which OS are you using? Windows?

amrsoll, 2020-01-09:
Is fixed in the plugin now

closed 2020-01-09 by amrsoll