mrbeam/OctoPrint #56: Ignore click into the work area

TopperDEL, 2015-11-23:
Clicking into the work area lets Mr.Beam move to that position. Sometimes I accidentially click because I wanted to move an object. This leads to unwanted movements. There could be a setting "Disable MoveByClick".

mrbeam, 2015-11-25:
Thanks for the issue. Some thoughts:
My primary goal is to not overload the UI with lots of switches and buttons. But I see that it is annoying to move the head unintentionally. 
I am always in doubt about the best possibility to move the Mr Beam. 
I think arrow buttons are annoying compared to the fast positioning by clicking right next to the placed object. 
The problem occurs if the object has just very thin borders that clicking on it is hard. In that case it moves unintentionally. To move skinny objects I added the 4-arrow-button in the list of placed objects. I confess this is quite subtle there, but for now I did not find a better solution. 
Some questions:
- Do you move the laser head for better positioning of the material on your working area? How often? 
- Do you move it unintentionally because the design you are trying to move has tiny lines? 
- What in detail is annoying about this unintentional move? 
- Did you recognize the 4-arrow-button in the list? 
- Would it be a solution to interpret clicks on the design more tolerant? e.g. clicking close to the line means clicking on it?

TopperDEL, 2015-11-25:
I think the best way would be to provide a button next to the 4-arrow-buttons saying "move head by clicking" or similar. That feature is nice, but a precise move is impossible with it. I would more often move the head with the 4-arrow-buttons instead as this gives me much more control.

On my last cut I've used that feature to show me the borders of my physical object within the software so I could place my design accordingly. But there the red cross was more helpful than the "move by click"-function.

mrbeam, 2015-11-25:
Still thinking about this. I use the "rapid positioning by click" quite
often.

What do you think about a "snap design to the red cross feature"?
ᐧ

closed 2023-05-30 by TopperDEL