mrbeam/OctoPrint #58: Firefox SVG rendering

mrbeam, 2015-12-08:
is crashing because of 'probably' this bug: https://bugzilla.mozilla.org/show_bug.cgi?id=700533

'''
source.onload = function () {
			renderCanvasContext.drawImage(source, 0, 0, renderCanvas.width, renderCanvas.height);

			// place fill bitmap into svg
			var fillBitmap = renderCanvas.toDataURL("image/png");
			if(typeof callback === 'function'){
				callback(fillBitmap);
			}
			renderCanvas.remove();
		};

		// catch browsers without native svg support
		source.onerror = function() {
			console.error("Can't export! Maybe your browser doesn't support native SVG. Sorry.");
		};
	};
'''
is running into the onerror case. 

TODO: check if svg width and height is properly set and read the bug report for possible workarounds.

make-ing, 2016-06-02:
Can't reproduce this bug. Need more information.

make-ing, 2016-06-16:
ok i can reproduce it now... It only happens for bitmaps like jpg and png.