mrbeam/OctoPrint #62: Not loading after upgrade

cojarbi, 2016-01-29:
Once connected to the Internet, octoprint shows various updates.  Once done it seems configuration is lost. Is this the correct process to upgrade or there I another proper way?

make-ing, 2016-01-29:
Hello Ariel, can you tell me which configuration is lost?

cojarbi, 2016-01-29:
@florian. After the update I can not longer connect to the server. It gives me a 503 error. I can ping the nic so I know the OS is up. 

Mind that there where 4 upgrades available octoprint, svg and other 2

Ariel Yahni

make-ing, 2016-01-29:
Have your tried to do a restart? I never had such a problem and also heard of this the first time.

cojarbi, 2016-01-29:
Yes. Restarted many times. Is the root user and password the default for octoprint. I want to ssh and sea what's going on. 

I had to reflash the card with the image and start all over bacause of this and did the process all over again with the same result. 

I saw that after upgrading the error bobble on the right that said octoprint failed to restart after the upgrade 

Ariel Yahni

cojarbi, 2016-02-01:
What's the correct way to update

make-ing, 2016-06-02:
@cojarbi: Is this problem still unsolved? The correct way would be to click on the "Update Now" if the "Update Available" dialog pops up.

cojarbi, 2016-06-02:
I have done that many times but haven't doesn't work. Also i get a constsnt lock as I don't have endstops and im unable to relsease

AY

make-ing, 2016-06-02:
The release Problem without homing is solved. See https://github.com/mrbeam/OctoPrint/issues/65
It is currently not in the stable-1.2.2 branch but in the development mrbeam branch.

Can you tell me your exact setup because i can not reproduce this upgrade behavior on my side.

cojarbi, 2016-06-02:
Will do, currently testing other firmware so i need to get back to that one

make-ing, 2016-06-22:
What setup do you have (RPi version, host OS, browser, network connection)?

closed 2017-02-08 by cojarbi