mrbeam/OctoPrint #63: [Request] Ability to place a part multiple times in the working area

Quasimondo, 2016-04-14:
Not sure if I am missing a hidden button, but right now it seems like I can place a part only once in the working area since once it is placed the "Add" button is disabled. It would be nice to have the option place the same element multiple times. Even better if there was a small interface that allows to place a part n times with an x or y offset (which is automatically filled with the bounds + 1mm or so)

make-ing, 2016-04-19:
We will work on it... Thanks for your request.

make-ing, 2016-06-16:
added functionality of multiple parts at once to a separate branch with commit: 74e77f7492f19d236a15e2f1d316aa4efd8118f2

Not sure when it will hit the stable branch cuz we have to check with additional changes in the same area.