mrbeam/OctoPrint #64: Change title of update notification

Gallore, 2016-05-03:
Current title says "Update Available". Change it to "Mr Beam Update Available" or something similar so that it's clear for the user that the Update is for Mr Beam. Internet users are facing fake update notifications on a regular basis. Make sure that the user understands that this update notification is about Mr Beam and that it's authentic.
[image]

Gallore, 2016-05-03:
[image]

make-ing, 2016-05-04:
Thank you very much Gallore for this suggestion... I will make the changes with the next update

make-ing, 2016-06-16:
This enhancement will come to stable with the next update.

closed 2016-06-16 by make-ing