mrbeam/OctoPrint #68: Not Converting

michizzo, 2016-06-16:
Hi Community,

my Mr. Beam Senior doesn't react after clicking the "laser" button.
The converting window appears and nothing happens. Even the Convert Button is not clickable.

What can I try to fix?

Thanks for answers and best regards Michel

make-ing, 2016-06-17:
Hello Michel,
can you give me a little bit more information?

-Are you using a standard senior kit?
-Which Browser are you using?
-Which Version are you using (you can find it on the bottom left of your webbrowser)?
-What filetype are you using?

Thanks Florian

michizzo, 2016-06-19:
Hi Florian,

Ja es ist ein Standard-Senior Kit 
Benutze einen iMac mit Safari.
Benutze die letzte Version, habe alles über das Netzwerk verbunden und somit auch die automatischen Fragen nach updates bestätigt. Auch bei manuellem Abfragen eines neuen updates ist alles aktuell.
Zum Lasern benutze ich .svg mit Illustrator herausgeschrieben (gleiche Datei lief auf dem gleichen Rechner mit den vorherigen Versionen).

Gruß Michel

make-ing, 2016-06-20:
Hi Michel,
am besten du verwendest Chrome als Browser fuer den Mr Beam. Leider haben wir nicht genuegend Zeit um uns um die grosse Anzahl verschiedener Browser zu kuemmern, und Chrome ist auf allen aktuellen Systemen verfuegbar.

Bitte geb mir bescheid ob es mit Chrome funktioniert, dann wuerde ich das Issue als Bug markieren aber erstmal auf die Seite legen bis wir zeit haben uns um Safari zu kuemmern. Waere das ok fuer dich?

Gruesse Florian

michizzo, 2016-06-20:
Hi Florian,

Natürlich ist das in Ordnung. Werde es direkt heute ausprobieren.

Gruß Michel