mrbeam/OctoPrint #95: Network connection lost and not possible to connect again.

CutAndCarve, 2017-12-17:
I noticed when my camera failed to refresh the picture of the workspace. The connection (which was working fine for hours before) was lost and I am not able to reconnect. I tried shutting down the Mr. Beam and starting again. I connected to the Mr. Beam's own Network before trying to reconnect to my own Wifi but that also didn't work.

In the current status I am not able to work whatsoever.

CutAndCarve, 2017-12-17:
Addition: I can see my Mr. Beam 2 on the find.mr-beam.org but when I click on it the screen "beamOS ist starting" won't go away.
[image]
[image]

oli-bischoff, 2018-02-13:
same here. but i can reconnect with pc and MBII wired to router. but no connection with wlan and ipad. worked last week perfectly.