mrbeam/OctoPrint #96: Cutting job instead of engraving during camera calibration

notyettakenusername, 2018-06-28:
When preparing for the camera calibration, laser job is being set to cutting instead of engraving even though "Engrave only" option has been chosen. 
[image]

Gallore, 2018-07-10:
Moved to Mr Beam Plugin: https://github.com/mrbeam/MrBeamPlugin/issues/310

Gallore, 2018-07-10:
Please submit Mr beam Tickets here:
https://github.com/mrbeam/MrBeamPlugin/issues
thx

-- 
*Andy Werner*

Head of Software

phone: +49 89 30908481

email: andy@mr-beam.org

www.mr-beam.org

Mr Beam Lasers GmbH | Gollierstr. 70d 3.OG | 80339 Munich | Germany
<https://maps.google.com/?q=Gollierstr.+70d+3.OG+%7C+80339+Munich+%7C+Germany&entry=gmail&source=g>

Geschäftsführer: Teja Philipp | Handelsregister: Amtsgericht München, HRB
212375

closed 2018-07-10 by Gallore