mrbeam/OctoPrint #98: Transformation in <use> doesn't work right

GordonGuenther, 2018-09-12:
Hi,

I've struggle with some strange behaviour of transformations in <use> tags in the webfrontend of my mr.beam 2. The display works in the design library but as soon as i put my svg into the working area the transformations get messed up. Apparently the values of translations will be interpreted as millimeters instead of pixels, which is wrong if a viewBox is set.

Here is a stripped down svg example:
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink"
  version="1.1" width="297mm" height="210mm" viewBox="0 0 2970 2100">
  <defs>
    <path id="object1" style="stroke:black;stroke-width:1px;fill:none" d="M25,25h150v150h-150z" />
    <path id="object2" style="stroke:black;stroke-width:1px;fill:none" d="M-150,-50h300v100h-300z" />
    <path id="object3" style="stroke:black;stroke-width:1px;fill:none" d="M-50,-150h100l-50,300z" />
  </defs>
  <g>
    <use xlink:href="#object1" />
    <use xlink:href="#object2" transform="translate(175,250)" />
    <use xlink:href="#object3" transform="translate(175,375) rotate(90)" />
  </g>
</svg>
```

Because of the used viewBox (1mm equals 10px) the translations are 10 times as big as they should be in the working area. Strangely everything works as expected if i use the paths directly with the same transformations. The svg will work:
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink"
  version="1.1" width="297mm" height="210mm" viewBox="0 0 2970 2100">
  <g>
    <path id="object1" style="stroke:black;stroke-width:1px;fill:none" d="M25,25h150v150h-150z" />
    <path id="object2" style="stroke:black;stroke-width:1px;fill:none" d="M-150,-50h300v100h-300z" transform="translate(175,250)" />
    <path id="object3" style="stroke:black;stroke-width:1px;fill:none" d="M-50,-150h100l-50,300z" transform="translate(175,375) rotate(90)" />
  </g>
</svg>
```

I don't know if this issue only affects <use> and if it'is specific to the mr. beam or a general octoprint problem.

If you need more information or screenshots just let me know.

Regards
Gordon

GordonGuenther, 2018-09-12:
The issue seems to be even bigger, than i thought.

When i leave my objects within the `<devs>` section and define the transformation there, the svg will be rendered correctly in the preview of the working area, but the objects will be omitted on the cutting. So it seems to me that `<use>` tags are also not supported by the slicing process correctly.

hungerpirat, 2018-09-13:
Dear Gordon, thanks for reporting this. 
I assume it is a duplicate of https://github.com/mrbeam/MrBeamPlugin/pull/330 
There is already a fix, waiting to be published in an upcoming software update.

closed 2019-12-17 by amrsoll