mrbeam/OctoPrint #106: Bugs in the develop tier

amrsoll, 2021-05-18:
It was reported that the latest OctoPrint pre-release creates bugs in the classic develop Tier.

Josef-MrBeam, 2021-05-19:
2021-05-18 07:43:46,798 - octoprint.plugins.softwareupdate - INFO - Update of octoprint to v1.3.6.5 successful!
2021-05-18 07:43:46,894 - octoprint.plugins.softwareupdate - INFO - Starting update of mrbeam to b05f3d43cf4647b6bd943ca3251464a9b3505d09...
2021-05-18 07:44:25,097 - octoprint.plugins.softwareupdate - INFO - Update of mrbeam to b05f3d43cf4647b6bd943ca3251464a9b3505d09 successful!
2021-05-18 07:44:25,347 - octoprint.plugins.softwareupdate - INFO - Starting update of mrbeam-ledstrips to db6bdad6826ff344aa889f01d8eef7d8663040f6...
2021-05-18 07:44:26,790 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2021-05-18 07:44:26,832 - octoprint.util.pip - ERROR - Error while trying to install testballoon to figure out pip setup
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/pip.py", line 349, in _check_pip_setup
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 1411, in run
    p.run(input=input, async=async)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 1040, in run
    self.run_node(node, input=input, async=async)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 1136, in run_node
    result = getattr(self, method)(node, input, async)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 1282, in run_command_node
    node.cmd.run(input=input, async=async)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 632, in run
    self.process = p = Popen(self.args, **self.kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 545, in _execute_child
    super(Popen, self)._execute_child(args, executable, preexec, *rest)
  File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory: '/home/pi/oprint/lib/python2.7/site-packages/octoprint/util/piptestballoon'
2021-05-18 07:44:26,838 - octoprint.util.pip - ERROR - Cannot use pip
2021-05-18 07:44:26,838 - octoprint.plugins.softwareupdate - ERROR - Update of mrbeam-ledstrips can not be performed
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 908, in _perform_update
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 79, in perform_update
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/pip.py", line 153, in execute
    output = p.stdout.text
UnknownPip
2021-05-18 07:45:19,826 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Github rate limit: 59/60, reset at 2021-05-18 08:45
2021-05-18 07:45:19,827 - octoprint.plugins.softwareupdate.version_checks.github_commit - DEBUG - Github rate limit: 58/60, reset at 2021-05-18 08:45
2021-05-18 07:45:19,831 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Target: octoprint, local: v1.3.6.4, remote: v1.3.6.5
2021-05-18 07:45:19,833 - octoprint.plugins.softwareupdate.version_checks.github_commit - DEBUG - Target: mrbeam, local: b1fa8c9639541af11c132634dba30c71ffb45bfb, remote: 60ec884a75f80c22383d4022b965a261db1d81f6
2021-05-18 07:45:19,932 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2021-05-18 07:45:27,323 - octoprint - ERROR - Exception on / [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/views.py", line 407, in index
    ctx = _request_ctx_stack.top
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/views.py", line 345, in plugin_view
    if "suffix" in data:
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 869, in decorated_function
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 923, in decorated_function
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 643, in decorated_function
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 625, in f_with_duration
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 878, in decorated_function
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 894, in decorated_function
  File "/home/pi/MrBeamPlugin/octoprint_mrbeam/__init__.py", line 803, in on_ui_render
[... cut ...]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_findmymrbeam/__init__.py", line 311, in _start_update_thread
    on_condition_false=self._on_disabled)
TypeError: __init__() got an unexpected keyword argument 'on_condition_false'
2021-05-18 07:53:21,932 - octoprint.plugins.netconnectd - INFO - Netconnectd status: ip_eth0: 192.168.1.48, ip_wlan0: 192.168.1.43, status: {u'connections': {u'ap': False, u'wifi': True, u'wired': True}, u'devs': [u'wlan0', u'eth0'], u'wifi': {u'valid_config': True, u'current_address': u'74:42:7F:B1:58:98', u'present': True, u'current_ssid': u'Bobafat'}, u'link': True}

amrsoll, 2021-05-19:
@Josef-MrBeam Please share a file next time instead of pasting hundreds of log lines